SpiNNakerManchester/JavaSpiNNaker #22: Use InetAddress instead of String whereever possible.

Christian-B, 2018-08-23:
For example in uk.ac.manchester.spinnaker.transceiver.UDPTransceiver.ConnectionFactor

uk.ac.manchester.spinnaker.connections.UDPConnection

We currently have 
uk.ac.manchester.spinnaker.machine.tags.IPTag using InetAddress

Which we have to convert back to a String to convert back to a InetAddress address.

If you agree you can assign back to me to change!

But do we want to use.
InetAddress or Inet4Address

I would be tempted to say that if they MUST be Inet4Address then lets use that!

dkfellows, 2018-08-28:
Seems reasonable to me. Mostly I've use `InetAddress` in public APIs because that's what Java's public APIs mostly provide back officially so it lets me avoid putting ugly casts in (and you can't ask for an instance of `Inet4Address` directly), but there's a clear argument in favour of `Inet4Address` too because SpiNNaker only handles IPv4 and that's not about to change. (IPv6 is quite a lot more complicated, to the point where it'd make SCAMP not fit on chip any more.)

rowleya, 2018-08-28:
I would say use InetAddress and use the fact that specifying an IPv6 address will result in no connectivity in any case!  If you really want to, you can always check inside the class, but this is unlikely to be necessary.  Think of specifying an IPv6 address on the same lines as specifying an incorrect but valid IPv4 address - it is a valid address but not the address of a SpiNNaker board.

Christian-B, 2018-08-28:
If spinnaker 2 is not planning on allowing Inet6 why pretend to support it?

As a big fan of fail early I prefer to use Inet4 and do the Class cast as early as possible.

I will create an InetFactory class to do this cleanly in one place.

if there are any other checks on what is a valid spinnaker Inet address these could be added here too.

rowleya, 2018-08-28:
Any IPv4 address is currently valid.  Note that it will fail pretty early if you specify an invalid address (similar to specifying an incorrect IPv4 address or host name), but I am happy enough to have our own Inet4Address static creation method if that is preferred.

I can't comment on whether SpiNNaker 2 will support IPv6 or not; I believe that the UDP protocol is a hardware layer, so it is possible that it could.  An option is to use InetAddress but have a verification static creation method in any case that returns InetAddress, but currently verifies that it is actually an Inet4Address.  This would save the need to update a load of classes if we find that we get IPv6 for free.

closed 2018-09-05 by dkfellows