SpiNNakerManchester/JavaSpiNNaker #23: Who should insert null default values.

Christian-B, 2018-08-23:
uk.ac.manchester.spinnaker.transceiver.UDPTransceiver.ConnectionFactory methods have only one or two parameters.

But classes like uk.ac.manchester.spinnaker.connections.EIEIOConnection require 4 parameters.

Would you agree that it is better to add default constructors that have the factory call methods with nulls?

If yyou agree assign back to me to change as required.

dkfellows, 2018-08-28:
Please feel free to add the constructors you think are required.

The `EIEIOConnection` in SpiNNMan has no `__init__` and the parent class `UDPConnection` allows all of `local_host`, `local_port`, `remote_host`, and `remote_port` to be defaulted (with `None`/`null` as default). Because of the different ways arguments and overloading work in Java, I've just put in the maximally specified constructors by default, and then added in others depending on what appears necessary at the time.

rowleya, 2018-08-28:
In general, local_port is needed for listening connections and local_host might then be specified as an additional thing if a specific host is to be used for listening on.  The use of remote_host and remote_port is for a remote connection, so generally both are needed for that.  So the combination (which is probably tricky to make work due to the types) is:
UDPConnection(int local_port)
UDPConnection(int local_port, InetAddress local_host)
UDPConnection(InetAddress remote_host, int remote_port)

I could easily have missed a case here...

Christian-B, 2018-10-01:
done

closed 2018-10-01 by Christian-B