SpiNNakerManchester/JavaSpiNNaker #75: ChipInformation seems redundant

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
the class ChipInformation.java in machine does not get used at all. 
What should use it (chipInfo in comms) does not use it. 
============================

What do you see that is wrong?
either redundant code, or incorrect instantiation of a class

What do you think it should do instead?
either move chipInformation or make chipInfo implement it

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-11-30:
can be removed as far I am am concerned

dkfellows, 2018-11-30:
I think this was a class from the comms model that you thought was needed at one point and then discovered you didn't need.

Christian-B, 2018-11-30:
https://github.com/SpiNNakerManchester/JavaSpiNNaker/pull/97

closed 2018-11-30 by Christian-B