SpiNNakerManchester/JavaSpiNNaker #77: none named magic numebrs

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
link.java hash code has magic numbers in it
============================

What do you see that is wrong?
they not named. I know this isnt major, but ya use named magic in tag.java hashcode, so keeping consistency, the link should have them named.

What do you think it should do instead?
name them

What version of our software are you using?
master
-------------------------------------------

alan-stokes, 2018-11-29:
also MachineDimenions.java hashCode

alan-stokes, 2018-11-29:
Processor.java

Christian-B, 2018-11-30:
hashcode methods are excempt from the magic number rule.

As defined in 
https://github.com/SpiNNakerManchester/JavaSpiNNaker/blob/master/src/support/checkstyle/style.xml

		<module name="MagicNumber">
			<property name="ignoreHashCodeMethod" value="true" />
		</module>

closed 2018-11-30 by Christian-B

alan-stokes, 2018-11-30:
lol, then why did ya name them for one of the classes? consistency

Christian-B, 2018-11-30:
We did not know we could turn it off when we did that.

Not worth changing back.

alan-stokes, 2018-11-30:
hum bug.