SpiNNakerManchester/JavaSpiNNaker #80: abnormal vs dead

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
machine.java has concepts of abnormal chips and links. Both of which look like to be the equiv of dead links and chips from the python. for consistency sake, these should be renamed to dead
============================

What do you see that is wrong?
inconsistenacy

What do you think it should do instead?
make consistent in one way or another

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-11-30:
No Abnormal is more than just dead.   For example a link is abnormal if it is only in one direction and a chip is abnormal if it has no links.

closed 2018-11-30 by Christian-B

alan-stokes, 2018-11-30:
a link working in one direction is a dead link in that direction. don't think dead links are considered bi directional. chip with no links should never happen, and i think the python deals with that before the java gets to it.....

Christian-B, 2018-11-30:
The Java code where possible should work without the python.
This is one of those cases.

Christian-B, 2018-11-30:
We have choose that is a link is dead in one direction it is "abnormal" in both!

alan-stokes, 2018-11-30:
bah, ill let you have this one.