SpiNNakerManchester/JavaSpiNNaker #83: MachineVersion.java

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
a bunch of random machine sizes 
 /** Combination of 3 common style 48 chips boards in a Toroid. */
    THREE_BOARD(12, 12),
    /** Combination of 24 common style 48 chips boards. */
    TWENTYFOUR_BOARD(48, 24),
    /** Combination of 120 common style 48 chips boards. */
    ONE_TWENTY_BOARD(96, 60),
    /** Combination of 600 common style 48 chips boards. */
    SIX_HUNDRED_BOARD(240, 120),
    /** Assumed Combination of 1200 common style 48 chips boards. */
    ONE_THOUSAND_TWO_HUNDRED_BOARD(240, 240),
============================

What do you see that is wrong?
well these are only ever used by java tests, and if your going to declaire these, your missing a load of combinations of machine sizes inbetween. These shouldnt be within the MachineVersion enum. You support building a enum with random values, so they should be built by the tests. Esp as nothign else in this class accesses these. 

What do you think it should do instead?
make tests build the enum they need, not plonkling them within the operating class.

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-11-30:
I choose to have a specific enum for each of the physical board combination that do or have existed in the real world.

May be overkill but leaving for now.

alan-stokes, 2018-11-30:
ave existed in the real world is already a incorrect argument. think spalloc allocating sets of triads, its not just 3, 24, 120, 600 (there's also 6 [which was built for the original column run] and the triad could do 9 12 etc..). so i think having those specific enums are dubious, esp given they are not atomic, whereas 2,3,4,5 are atomic things, they represent specific board formats