SpiNNakerManchester/JavaSpiNNaker #85: public vs private variables

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
regionLocation has public x,y,p,region, hashcode variables. but provides getX, getY getP hashCode methods. 
============================

What do you see that is wrong?
public variables which should be private

What do you think it should do instead?
make private, give a getter for region and mod the eq and compareto to use said getters. 

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-11-30:
final inmutables can be public. No reason not to.

closed 2018-11-30 by Christian-B

Christian-B, 2018-11-30:
in this case the getters are required by the interface.

alan-stokes, 2018-11-30:
that kinda makes my point. if your going to have yourself forced to have those getters, you might as well suppor tthat as the general route into the classes data. instead of 2 ways. will add confusion down line when you start accessing them in interchangable ways.

Christian-B, 2018-11-30:
The forced getters is because there are multiple implementations of this interface!

alan-stokes, 2018-11-30:
yes, but that is irrespective when considered in the use case of this class. this clas has 2 ways of accessing the same data. i remind you of python "1 million ways to do the same thing argument"

alan-stokes, 2018-11-30:
hum bug