SpiNNakerManchester/JavaSpiNNaker #86: style cripe

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
    public Router(HasChipLocation source, int clockSpeed,
            int nAvailableMulticastEntries,
            Collection<Direction> ignoreDirections, Machine machine) {
============================

What do you see that is wrong?

of these ways of doing things:
    public Router(HasChipLocation source, int clockSpeed,
            int nAvailableMulticastEntries,
            Collection<Direction> ignoreDirections, Machine machine) {

    public Router(
            HasChipLocation source, int clockSpeed,
            int nAvailableMulticastEntries,
            Collection<Direction> ignoreDirections, Machine machine) {

    public Router(HasChipLocation source, int clockSpeed,
                          int nAvailableMulticastEntries,
                          Collection<Direction> ignoreDirections, Machine machine) {
I find the first really annoying to read. 

What do you think it should do instead?

either 2 or 3, but avoid 1 

What version of our software are you using?
master
-------------------------------------------

alan-stokes, 2018-11-30:
the issue has now been formatted better. the issue is the mismatch of arg aligning. i find the one where the args start in the bracket, then move to 1 tab in from the definition very distracting. 

having them either aligned to the open bracket (as 3), or algined to the 1 tab from the get go (2) much more easy to read

dkfellows, 2018-11-30:
I disagree with you.

alan-stokes, 2018-11-30:
maybe we should have a authority dictate what we're going to accept as acceptable, instead of us hitting our individual drums? *looks towards @rowleya *

rowleya, 2018-11-30:
My opinion is to consult a style guide on this if it really bothers you.  However, I have always used 1 in Java...

rowleya, 2018-11-30:
https://www.oracle.com/technetwork/java/javase/documentation/codeconventions-136091.html#248

alan-stokes, 2018-11-30:
well if @rowleya uses 1. i guess ill throw my drum down and just grumble to myself. but ill be avoiding that style in java code i write.

closed 2018-11-30 by Christian-B