SpiNNakerManchester/JavaSpiNNaker #87: magic number and odd result

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
RoutingEntry has a magic number in figuring NUM_PROCESSORS (aka 32) and to add, that calculation returns a result of 26, which at last check, was not how many cores a spinnaker 1 has
============================

What do you see that is wrong?
magic num and wrong resulting value

What do you think it should do instead?
should be 18 or MAX_NUM_CORES from MachineDefaults.java

What version of our software are you using?
master
-------------------------------------------

alan-stokes, 2018-11-29:
i dont think this has actually caused any issues, as the cores from machine shouldnt never go above 18. but its a opening for dubious.

alan-stokes, 2018-11-30:
but 8 bits of that are not used by cores. not sure what those 8 bits are used for. but we obvs dont have cores 19 to 26

dkfellows, 2018-12-03:
Had to hunt around: in `spinnaker_tools/sark/sark_hw.c` in line 302, you can see that the top eight bits are used for the `app_id`. Without masking.

We'd better make sure that those bits are zero when setting them in here.

Christian-B, 2021-03-22:
@dkfellows is this fixed now?

dkfellows, 2021-03-25:
Looks fixed to me; I can't see any way to use out-of-range processor IDs now.

closed 2021-03-25 by dkfellows