SpiNNakerManchester/JavaSpiNNaker #88: public vs private

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
SpiNNakerTriadGeometry.java has 2 public variables that are not used anywhere outside of the class. 
    /** Height of a triad in chips. */
    public final int triadHeight;
    /** Width of a triad in chips. */
    public final int triadWidth;
============================

What do you see that is wrong?

    /** Height of a triad in chips. */
    public final int triadHeight;
    /** Width of a triad in chips. */
    public final int triadWidth;

What do you think it should do instead?

    /** Height of a triad in chips. */
    private final int triadHeight;
    /** Width of a triad in chips. */
    private final int triadWidth;

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-11-30:
final imutable can be public

closed 2018-11-30 by Christian-B

alan-stokes, 2018-11-30:
but theres no need in this case. nothing would use them outside this classes internals

Christian-B, 2018-11-30:
no need != bad

alan-stokes, 2018-11-30:
lots of little no need != bad can accum to == bad. lol.