SpiNNakerManchester/JavaSpiNNaker #89: 80 lines limit (assuming we're sticking to this

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
TriadGeo.java line 86
============================

What do you see that is wrong?
over 80 chars

What do you think it should do instead?
not be

What version of our software are you using?
master
-------------------------------------------

dkfellows, 2018-11-30:
Imports are excepted from the 80 column rule. I advise telling your IDE to automatically fold the imports out of sight anyway; they're usually very very very boring.

alan-stokes, 2018-11-30:
it does. I open them when im having a gander for things to find. as the ide seems a bit....... pinickity to when itll auto to declarations and not. 

but there is a lot of places where the code itself goes over 80 chars.

dkfellows, 2018-11-30:
There are no places where the code goes over 80 characters. That's an auto-enforced rule.

(Tabs are width 4.)

alan-stokes, 2018-11-30:
i was going to argue. till i scrolled through a lot of the examples i was going to use, and noticed that the ideaj had automatically added variables names next to calls, so they looked like they went over 80 chars, but in reality, those are not actually code. grrgle. 

I'll trust ya that its woriking and its just my ide

Christian-B, 2018-11-30:
Alan Trusts us. Close quickly!

closed 2018-11-30 by Christian-B