SpiNNakerManchester/JavaSpiNNaker #90: too far indented

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
triad geo.java
============================

What do you see that is wrong?
lines 102 and 105 are indented by 2 tabs, should be 4 spaces (aka 1 tab)
line 139 isnt aligned properly
lines 167 and 168 wrong indentation
line 195 199 200 247 249

What do you think it should do instead?
 should be 4 spaces (aka 1 tab)

What version of our software are you using?
master
-------------------------------------------

alan-stokes, 2018-11-29:
command line interface.java line 85, 68,69, 
48 (or at least it needs a extra space to align it)

alan-stokes, 2018-11-29:
StartSendingMessages line 52

alan-stokes, 2018-11-29:
ProtocolMessage line 47

alan-stokes, 2018-11-29:
MissingSeqNUmMessage line 44 83 84 106 107

alan-stokes, 2018-11-29:
HostDataREceiver lines 213 245 260 266 290 291 296 301 339

alan-stokes, 2018-11-29:
DataOut lines 158 through to 190, 227 and 228 273 and 274

alan-stokes, 2018-11-29:
Constants.java in front end. line 38

alan-stokes, 2018-11-30:
memoryRegionCollection line 91

alan-stokes, 2018-11-30:
Functions.java line 167 168 169 170 180 181 182 205 206 207 208 209 229 243 244272 273 296 297 298 341

alan-stokes, 2018-11-30:
UnimplementedDSE command exception line 39

alan-stokes, 2018-11-30:
test data spec executor.java lines 167 168 169 170

alan-stokes, 2018-11-30:
test memory region collection.java lines 38 40 61 64 67 83 86 88

alan-stokes, 2018-11-30:
test operation mapper lines 49 51 63 75 88 99 100 112 113 130

Christian-B, 2018-11-30:
StartSendingMessages  What where? ignoring!

Christian-B, 2018-11-30:
MissingSeqNUmMessage ??? Ignoring!

dkfellows, 2018-11-30:
I think continuation lines should usually be done with a double-indent.

Christian-B, 2018-11-30:
generator.java line 191
That reads better WITH 2 tabs/ 8 chars 
NOT changing

Christian-B, 2018-11-30:
while I agree that even tests should be generally well styled don't care enough to change

Christian-B, 2018-11-30:
Don't care but we have to give the Scots some victories.
https://github.com/SpiNNakerManchester/JavaSpiNNaker/pull/97

alan-stokes, 2018-12-04:
this happens a lot in comms. 8 spaces vs 4. example line 1693 in txrx

alan-stokes, 2018-12-04:
StartSendingMessages  = StartSendingMessage
MissingSeqNUmMessage  = MissingSequenceNumbersMessage

dkfellows, 2018-12-06:
I cannot find the code that you are talking about.

alan-stokes, 2018-12-06:
this was agreed that it was to be 4 spaces. not 8. as pointed out above by @Christian-B  

"Don't care but we have to give the Scots some victories.
#97"