SpiNNakerManchester/JavaSpiNNaker #93: try without catch

alan-stokes, 2018-11-29:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
HostDataRerceiver.java lines 200 to 203. has no catch
============================

What do you see that is wrong?
has no catch

What do you think it should do instead?
have a catch of some level, even its just a overall exception

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-11-30:
https://docs.oracle.com/javase/tutorial/essential/exceptions/tryResourceClose.html

closed 2018-11-30 by Christian-B

alan-stokes, 2018-11-30:
fair enough. *I bow to your superiority here*

Christian-B, 2018-11-30:
Where is that print button!

dkfellows, 2018-11-30:
Chimp beats puppy.

alan-stokes, 2018-11-30:
lol, chimp beats sick puppy. chimp beats puppy when chimp is right.