SpiNNakerManchester/JavaSpiNNaker #95: dse reload functionality using a constant that is not consistent between machines

alan-stokes, 2018-11-30:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
private static final int DEFAULT_SDRAM_BYTES = 117 * 1024 * 1024;
============================

What do you see that is wrong?
this constant is possible to not be a constant and should be read from the machine chip in question

What do you think it should do instead?
public void reload(HasCoreLocation core, LocationInfo info, File dataSpec) should also take the machine, and read the chip linked to coreLocation and get the sdram from the specific chip. 

What version of our software are you using?
master
-------------------------------------------

alan-stokes, 2018-11-30:
and here
public void reload(HasCoreLocation core, File dataSpec)

Christian-B, 2018-11-30:
These methods have neither a Chip nor Machine object from which to get this info.

Christian-B, 2018-11-30:
The constant however if used should be in MachineDefaults.

dkfellows, 2018-11-30:
I'm more likely to kill that method entirely sometime in the future.

alan-stokes, 2018-11-30:
up to you. i think the changes are relatively easy. and reload i think reflects the run, set, run mode. so it might be worth having the support in java for parallel support. but up to ya

dkfellows, 2021-03-25:
* The methods concerned are gone, so fixing them is impossible.
* The constant is in `uk.ac.manchester.spinnaker.machine.MachineDefaults`.

closed 2021-03-25 by dkfellows