SpiNNakerManchester/JavaSpiNNaker #101: comments and code and new lines

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

This is more a place holder for us to dicuss. not really a wrong or right. 
there are currently 2 conventions happening in the java code.

/** comment*/
code
/** comment*/
code

and

/** comment*/
code

/** comment*/
code

(note the new line between new blocks). Persoanlly i find the second more readable than the first, and most of the java ive seen to date does the second, but the comms seems to follow the first. 

discuss if we want a unified convention here, or meh to either. Im not sure, so leaving it as issue for opening debate

Christian-B, 2018-12-03:
There is no known way of enforcing this so hard to see this as an issue.

Personally I think the line break should be used to break code into logical block irrespective if there are comments in the code or not.

alan-stokes, 2018-12-03:
that's why it was a question flag. And id prefer the splitting, but because its not testable, we need to agree as a team. thus why i opened it here, so that the rest who are not in the office, can put their 2 pence in

rowleya, 2018-12-03:
The Java coding conventions say that a single line comment must be proceeded by a blank line:
https://www.oracle.com/technetwork/java/codeconventions-141999.html

Note that I don't care personally if we follow these or not, but our funders might (eventually) want us to be able to point at a document of what conventions we are using.  Thus we can either write one or use one.  At that point, I prefer to use one...

Whether it is checkable is a different issue - I suspect there will be a way to get checkstyle to check something like this, though I admit that I am not sure how!

dkfellows, 2018-12-03:
http://checkstyle.sourceforge.net/config_whitespace.html#EmptyLineSeparator

Note that introducing a new checkstyle rule is an intrusive thing. It's best done in a branch where the first commit is _just_ the change to the style rule.

alan-stokes, 2018-12-03:
ok, so then turning this into a minor issue for a pr to link to.

alan-stokes, 2020-07-08:
looks like #207 makes this mute. so closing

closed 2020-07-08 by alan-stokes