SpiNNakerManchester/JavaSpiNNaker #103: // force line wrap

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
	private final Map<BMPCoords, ConnectionSelector<BMPConnection>>
		// force line wrap
		bmpSelectors = new HashMap<>();
============================

What do you see that is wrong?
i find this very disconnecting. Ive been hit by it before with other modules, but id rather have a better solution than this

What do you think it should do instead?
can it be done as follows:
	private final Map<BMPCoords, ConnectionSelector<BMPConnection>>
		bmpSelectors = new HashMap<>(); // force line wrap

What version of our software are you using?
master
-------------------------------------------

rowleya, 2018-12-03:
Hmm, I agree that this uglifies the code for a formatting issue.  My eclipse formatter does this with the line:
```
private final Map<BMPCoords, ConnectionSelector> bmpSelectors =
            new HashMap<>();
```
I am happy with this.  Is there something worse that eclipse does?

rowleya, 2018-12-03:
I should add that the above is less ugly in my opinion than
```
private final Map<BMPCoords, ConnectionSelector>
// force line wrap
bmpSelectors = new HashMap<>();
```

dkfellows, 2018-12-03:
The problem is that this:

	private final Map<BMPCoords, ConnectionSelector<BMPConnection>> bmpSelectors =
			new HashMap<>();

exceeds the maximum line length (by 2 characters). I need to figure out how to allow Eclipse to auto-wrap in other places in the line.

rowleya, 2018-12-03:
Mine produces this:
```
private final Map<BMPCoords,
            ConnectionSelector<BMPConnection>> bmpSelectors =
                    new HashMap<>();
```

Does yours not let this happen?

rowleya, 2018-12-03:
Here are my rules:
[JavaFormatter.xml.txt](https://github.com/SpiNNakerManchester/JavaSpiNNaker/files/2639577/JavaFormatter.xml.txt)

I must have updated them from the defaults, but admit that I don't know what I changed!  One minor change has resulted in this:
```
private final Map<BMPCoords,
            ConnectionSelector<BMPConnection>> bmpSelectors = new HashMap<>();
```

dkfellows, 2018-12-03:
Found the setting. Obscure? You betcha!

alan-stokes, 2018-12-03:
there's at least 1 more place i remember seeing this. might be worth doing a search for the force line thingy.

dkfellows, 2018-12-06:
Must've fixed that in passing.

closed 2018-12-06 by dkfellows