SpiNNakerManchester/JavaSpiNNaker #104: _nearest_neighbour_id and _nearest_neighbour_lock in Transceiver.java

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
these are instance variables. But are only used by flood fill. 
============================

What do you see that is wrong?
Given we cant flood fill more than once at the same time, having a lock for this seems a bit odd. Unless there's issues between instances going back to 0, id think ya could push it to the method scope instead of instance level. 

What do you think it should do instead?
push to scope of flood fill method

What version of our software are you using?
master
-------------------------------------------

rowleya, 2018-12-03:
Given you can't flood-fill more than once at the same time, don't we need it to be an instance lock so you can't, actually, flood-fill more than once at the same time?

dkfellows, 2018-12-03:
Why do you say that we can't do more than one Java operation at a time?

alan-stokes, 2018-12-03:
ok, fine, so the lock might be fine, but the instance count variable?

rowleya, 2018-12-03:
Ah NN ids are global on SCAMP.  This avoids clashing NN packets.

alan-stokes, 2018-12-04:
but they cycle anyhows. and as we just discussed about only having 1 flood fill, and that id is only used in this one method. thus, why cant it start at 0 inside the method and iterate within itself, instead of having a class param?

rowleya, 2018-12-04:
I think this is because you could in theory have two transceiver instances pointed at the same machine.  Clearly only one should be able to update the NN id at the same time.  A class-level variable is the only way to protect this properly - the semantics are that no matter how many Transceivers you create, this shouldn't go wrong.

rowleya, 2018-12-04:
Although the NN ids wrap, you shouldn't reset the number close to the last time you used it as you don't know for sure that NN packets aren't still flying around.  At best, you can have an instance-level variable, but given the ability to create more than one transceiver, I still think keeping it as it is now is the best option all round.

Although there is a Java transceiver, is this ever used when the Python transceiver is in use?  I admit that it is hard to design for this scenario, but there is no need to make the system less reliable just because it can be done; there is also no way to stop a user opening ybug and sending in commands at the same time, but it isn't recommended!

alan-stokes, 2018-12-04:
from ya own statement:
 there is also no way to stop a user opening ybug and sending in commands at the same time, but it isn't recommended!

if ya follow this logic down, why even bother adding the safety for 2 python txrx's. Id be very suspicious of someone building a second one when the first will do. lol. maybe we should just add safety code to the constructor to stop multiple txrx's on the same board/ipaddress?

will java and python txrx be used at the same time ...... the statement above makes me want to say "yep, end users will do insane things" and this is only over flood fill......... which would be insane to do in both at the same time. lol. 

I think this:
Although the NN ids wrap, you shouldn't reset the number close to the last time you used it as you don't know for sure that NN packets aren't still flying around.

means we should leave it be though.... urg, how to handle both txrx's is going to add some complex stuff me thinks, if we really want them to handle each others behaviours.

rowleya, 2018-12-04:
OK, but why are you so bothered?  This prevents one way of doing this accidentally.  You can't stop every bad thing from happening sure, but you may as well add in safety where you can.  This doesn't break the code in any way - if anything it at least highlights the issue, so I don't think it needs to be changed.

alan-stokes, 2018-12-04:
it was just flagged when i was looking at the java transceiver and saw the odd variable, which then caused this train of thought (mostly from max_core that set this off). so given the logic and urg of getting it right. closing

closed 2018-12-04 by alan-stokes