SpiNNakerManchester/JavaSpiNNaker #105: transceiver constructors hosst== null issue

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
		if (host != null) {
			log.info("Creating transceiver for {}", host);
		}
============================

What do you see that is wrong?
that there is no boom boom when it is null. what will happen is ya get a listen connection for boot (even fi you define all connections) and then during boot itll crash out when trying to send. 

What do you think it should do instead?
check and go boom boom at constructor time

What version of our software are you using?
master
-------------------------------------------

rowleya, 2018-12-03:
The idea here (in SpiNNMan at least) was that could could pass a set of connections (including a boot connection) rather than a hostname.  Thus hostname being null is OK.  Not sure if this is still true in SpiNNMan, or if it is true here.

dkfellows, 2018-12-03:
@alan-stokes noticed that the boot connection insists on having a real hostname.

rowleya, 2018-12-03:
But if you give the Transceiver a BootConnection which has already been created, won't that work?  Note that I don't actually think we ever use the Transceiver like this, so enforcing the specification of the hostname might be useful...

closed 2018-12-03 by alan-stokes