SpiNNakerManchester/JavaSpiNNaker #112: Transceiver.java ExecuteLock

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
the class def for ExecuteLock is at lines 676 to 698, the usage of said class doesnt start till 1291.
============================

What do you see that is wrong?
the executeLock is nowhere near the code that uses it

What do you think it should do instead?
either:
1. be located next to where its used
2. plonk all these extra classes at the bottom of the class, so that we're not muddying the water

I personally perfer 2. 

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-12-03:
I have a preference for defining inner classes at the very bottom. (except anonymous ones of course)

I would accept if it is just above or just below the ONLY place the classes is used but still ugly!

dkfellows, 2018-12-06:
The whole locking mechanism is ugly.

PR #212 (2018-12-14): Yet another tranche of small fixes
fixes #112 
fixes #167 
fixes #168 
fixes #169 
fixes #200 
fixes #201 
fixes #202
fixes #203 
fixes #206
closed 2018-12-10 by dkfellows