SpiNNakerManchester/JavaSpiNNaker #113: Transceiver.java

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
calls not documented
============================

What do you see that is wrong?
calls e.g. getScampConnectionSelector has no java doc, but is a public method

What do you think it should do instead?
have javadoc

What version of our software are you using?
master
-------------------------------------------

dkfellows, 2018-12-06:
Thou art wrong. It's got Javadoc in the interface that specifies it.
> ConnectionSelector\<SCPConnection> uk.ac.manchester.spinnaker.transceiver.Transceiver.getScampConnectionSelector()
>
> @Override
> **Specified by:** getScampConnectionSelector() in TransceiverInterface
> **Returns:** The connection selector to use for SCP messages.

closed 2018-12-06 by dkfellows

alan-stokes, 2018-12-06:
fine. didnt get to that class before this one