SpiNNakerManchester/JavaSpiNNaker #114: Transceiver.java variables being declared in middle of class

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
block of magic variables in lines 722 to 733
Some of these are not even used till lines 1293 (e.g. EXECUTABLE_ADDRESS)
============================

What do you see that is wrong?
Either they are local variables for 1 method (like hashCode) or they are true class defs, which means they should be at the top. Like all sensible variable defs. 

What do you think it should do instead?
be moved to correct locations

What version of our software are you using?
master
-------------------------------------------

alan-stokes, 2018-12-03:
same for CONNECTION_CHECK_DELAY    (line 782)

its used in 2 methods (granted, those two methods are right next to each other, and right below the definition, but that's not 100% to be ensured during future proofing. its safer I think, when its in the top point when its used by more than 1 method

dkfellows, 2018-12-06:
You missed a bunch of constants. (`NNID_MAX`, `POST_BOOT_DELAY`, `POST_POWER_ON_DELAY`, `ENABLE_SHIFT`, `ROUTER_DIAGNOSTIC_COUNTER_ADDR`)

alan-stokes, 2018-12-06:
yea, but that was due to previous logic. but discussion states all go to top. so assume blanket arg for said change

closed 2018-12-06 by dkfellows