SpiNNakerManchester/JavaSpiNNaker #115: using chemical gas constant

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
line 798
============================

What do you see that is wrong?
turn into a true variable, retry comes to mind. not the gas constant. 

What do you think it should do instead?
better name

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-12-04:
Line 798 of WHAT!

alan-stokes, 2018-12-04:
Transceiver.java

dkfellows, 2018-12-05:
“chemical gas constant”??? Wat.

alan-stokes, 2018-12-05:
r 
basically 1 letter variables. retry instead of r. theres also conn/connection instead of c all over the place.

dkfellows, 2018-12-06:
Using a single-letter variable _internally_ to a method isn't a big deal.

alan-stokes, 2018-12-06:
readability. i reject your response and reopen

alan-stokes, 2018-12-06:
*pulls out the drums*

dkfellows, 2018-12-11:
FWIW, I've _seen_ what actually happens when people put in style rules to prohibit local variable names like `i` and so on…

Programmers mostly just double the letter up! Yes, `ii` is the new `i`. 🤦‍♂️

closed 2018-12-11 by dkfellows

alan-stokes, 2018-12-11:
Programmers mostly just double the letter up! Yes, ii is the new i

WTF! are you telling me youd do that! coz if so, my respect for ya just dropped a notch. lol.  i, or ii, or iii or any number of i's is not a valid variable name.

rowleya, 2018-12-11:
You could add a rule that requires that every variable is made up of dictionary words!

(For the avoidance of doubt, the above is not expected to be taken seriously).

Christian-B, 2018-12-13:
I can understand the python rule that disallows the single letter variable names l and O as they looks like the numbers 1 and 0.

But other single variable names are a question of taste.

For i ... is perfectly fine. 

We also have x, y, p all over the place and that is fine and should not be otherwise.

The c and r here are more open to taste,