SpiNNakerManchester/JavaSpiNNaker #118: Transceiver.java 932 and 795

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
check connection is only used by discoverScampConnections. The int retries thus can be removed and the magicNumber STANDARD_RETRIES_NO could be pushed inside instead
============================

What do you see that is wrong?
extra utilisation of magic number

What do you think it should do instead?
psuh STANDARD_RETRIES_NO into check_connection

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-12-04:
GIGO Python is the same.

alan-stokes, 2018-12-04:
so lets repeat the steps of issue https://github.com/SpiNNakerManchester/JavaSpiNNaker/issues/117
so to rowley we go

Christian-B, 2019-01-02:
Fixed in both Java and Python

closed 2019-01-02 by Christian-B