SpiNNakerManchester/JavaSpiNNaker #120: getSystemVariable

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
getSystemVariable can return a null, but code depending on it will automaitcally cast it to byte[]. so blow up
============================

What do you see that is wrong?
the default of the getSystemVariable should be a error instead of null

What do you think it should do instead?
error instead of null

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-12-04:
Transceiver Line 652:

Agreed this is sloppy coding.
Better to throw an UnsupportedOperationException

dkfellows, 2018-12-05:
IllegalStateException is better.

closed 2018-12-06 by dkfellows