SpiNNakerManchester/JavaSpiNNaker #99: trasnceiver.machine

alan-stokes, 2018-12-03:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
dont like it being package protected, should be private
============================

What do you see that is wrong?
 package protected, should be private

What do you think it should do instead?
access only at that level for test. should utilise back door

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-12-03:
Note: the correct fix is to use ugly reflection.

rowleya, 2018-12-03:
Using reflection here is probably the better answer.  Opening access to methods in a class just for testing seems like a bad idea...

Note the reflection code is quite simple:
```
Method method = targetClass.getDeclaredMethod(methodName, argClasses);
method.setAccessible(true);
method.invoke(null, argObjects);
```

rowleya, 2018-12-03:
Note that the alternative recommendation is to *not* test private methods directly, but rather indirectly through public methods.

Christian-B, 2018-12-03:
No here public methods are bad as outside testing you should never be able to set the machine in a transceiver directly!  Nor should a transceiver have a virtual Machine!

PR #100 (2018-12-03): Make the Transceiver's machine field private
Fixes #99
closed 2018-12-03 by alan-stokes