SpiNNakerManchester/JavaSpiNNaker #127: transceiver line 1218

alan-stokes, 2018-12-04:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
	@SuppressWarnings("deprecation")
============================

What do you see that is wrong?
whys this here?

What do you think it should do instead?
find a way to avoid having it

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-12-04:
I hate the use of this annotation.

it needs to be replaced by one of.
1. mark these methods also as depricated
2. Remove the derication tag from used inner method
3. Change the code to use a none depricated solution!

dkfellows, 2018-12-05:
I think there was a stage when it was deprecated because it used a deprecated bit of the machine model, which has subsequently become un-deprecated.

alan-stokes, 2018-12-05:
...... does that make this code un-deprecated then?

closed 2018-12-06 by dkfellows