SpiNNakerManchester/JavaSpiNNaker #129: txrx line 1296

alan-stokes, 2018-12-04:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
simpleProcess()
					.execute(new ApplicationRun(appID, chip, processors, wait));
		
============================

What do you see that is wrong?
it feels like a bad split there

What do you think it should do instead?
simpleProcess().execute(new ApplicationRun(
        appID, chip, processors, wait));
		

What version of our software are you using?
master
-------------------------------------------

alan-stokes, 2018-12-04:
same for line 1313

alan-stokes, 2018-12-04:
1328

alan-stokes, 2018-12-04:
1857

dkfellows, 2018-12-05:
I know. But I won't fix as this is where the Eclipse autoformatting rules I'm using pick, and I really don't want to fight those.

closed 2018-12-05 by dkfellows