SpiNNakerManchester/JavaSpiNNaker #131: txrx getConnectionList (line 616)

alan-stokes, 2018-12-04:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
this looks like our python trying to use 2 possible default params
============================

What do you see that is wrong?
 2 possible default params

What do you think it should do instead?
we can split this into board and connection and allow code that utilises it (aka line 1690 to not need to put null in the other field)
lines to change are 1690, 1726, 1742, 1751

What version of our software are you using?
master
-------------------------------------------

dkfellows, 2018-12-06:
Can't remove, as `clearIPTag` passes both arguments from its caller.

dkfellows, 2018-12-06:
We can and should slaughter the `connection` parameter from `Transceiver.clear_ip_tag`.

Christian-B, 2019-01-02:
fixed in both Java and python

closed 2019-01-02 by Christian-B