SpiNNakerManchester/JavaSpiNNaker #154: "machine", "chip_x", "chip_y", "where_is", "cabinet", "frame", "board", "x", "y", "z", "job_id"

alan-stokes, 2018-12-05:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
"machine", "chip_x", "chip_y", "where_is", "cabinet", "frame", "board", "x", "y", "z", "job_id"
these are used loads of places
============================

What do you see that is wrong?
currently magic strings

What do you think it should do instead?
constants to link to these

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-12-05:
If there is even a remote chance any of these could change I would argee.

For example in database column names.

But in these cases I don't think they will ever every change.

alan-stokes, 2018-12-05:
up to you two.

dkfellows, 2018-12-06:
I think those are all determined by the Spalloc protocol.

closed 2018-12-06 by dkfellows