SpiNNakerManchester/JavaSpiNNaker #156: JobsChangedNotification.java

alan-stokes, 2018-12-05:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
		setJobsChanged(asList(jobID)); (line 42)
============================

What do you see that is wrong?
asList expects a list of stuff, not a singleton. 

What do you think it should do instead?
		setJobsChanged(Collections.singletonList(jobID));

What version of our software are you using?
master
-------------------------------------------

Christian-B, 2018-12-05:
Arrays.asList works with all Object singletons.

it does not work on primitive singletons.

alan-stokes, 2018-12-05:
yes, but the ide complained that Collections.singletonList(jobID)); is clearer ;-)

Christian-B, 2018-12-05:
Collections.singletonList(jobID)) result s not mutable while Arrays.asList result is so they are not the same thing.

closed 2018-12-06 by dkfellows