SpiNNakerManchester/JavaSpiNNaker #162: Constants.java in messages module

alan-stokes, 2018-12-05:
_Remember: the better you describe your issue, the easier it is for us to help you._

What is the problem you see?
NO_MIDDLE_CHIPS_TO_CHECK never used (also never used in python, cull required)
DEFAULT_SDP_TAG is the same issue
EIEIO_DATA_HEADER_SIZE is the same
EIEIO_COMMAND_HEADER_SIZE is the same
SYSTEM_VARIABLE_BYTES is the same
============================

What do you see that is wrong?
these are not ANYWHERE

What do you think it should do instead?
go bye bye

What version of our software are you using?
master
-------------------------------------------

alan-stokes, 2018-12-05:
and SHORT_SIZE

dkfellows, 2018-12-06:
But how else will we know the size of shorts to wear?

alan-stokes, 2018-12-13:
so it was being used by a clone?

Christian-B, 2019-01-02:
removed python too label as fixed in 
https://github.com/SpiNNakerManchester/SpiNNMan/commit/ecc70d4472bce6664b96d08d4f56d721618ca831

Christian-B, 2019-01-02:
added wontfix as only remaining constant not being removed

dkfellows, 2021-03-25:
Remaining actions need not block 6.0 release

closed 2023-01-10 by dkfellows