SpiNNakerManchester/JavaSpiNNaker #170: SystemVariableDefinition.java

alan-stokes, 2018-12-06:
Full path to where you find the problem:
JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\messages\model\SystemVariableDefinition.java

What is the problem you see?
padding1,2,3,4 add comment to @Deprecated to why its deprecated (rowley advised)

What do you think it should do instead?

What version of our software are you using?

alan-stokes, 2018-12-06:
same on C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\messages\model\CPUState.java

for CPU_STATE_12 ,13, 14

Christian-B, 2018-12-06:
spinnaker/SpiNNMan/spinnman/model/enums/cpu_state.py

can you guarantee these will never be used?

dkfellows, 2018-12-06:
The padding values are deprecated because they're _just padding bytes_.

alan-stokes, 2018-12-06:
still add docs.

dkfellows, 2018-12-07:
No.

closed 2018-12-07 by dkfellows