SpiNNakerManchester/JavaSpiNNaker #172: ChipSummaryInfo.java (IDE gripe)

alan-stokes, 2018-12-06:
Full path to where you find the problem:
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\messages\model\ChipSummaryInfo.java

What is the problem you see?
private static final byte[] NO_ADDRESS = new byte[ADDRESS_SIZE];

What do you think it should do instead?
    @SuppressWarnings("MismatchedReadAndWriteOfArray")
    private static final byte[] NO_ADDRESS = new byte[ADDRESS_SIZE];

What version of our software are you using?
master

dkfellows, 2018-12-06:
I don't understand this warning.

alan-stokes, 2018-12-06:
the warning is that your building a array that you never write to. I guess the ide doesnt like arrays that are only ever read. choose what you wish with it.

dkfellows, 2018-12-07:
Wrote it out more explicitly.

closed 2018-12-07 by dkfellows