SpiNNakerManchester/JavaSpiNNaker #177: 0b00000011

alan-stokes, 2018-12-06:
Full path to where you find the problem:
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\messages\bmp\ReadFPGARegister.java
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\messages\bmp\WriteFPGARegister.java
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\messages\eieio\EIEIODataMessage.java
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\messages\scp\FloodFillStart.java
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-data-specification\src\main\java\uk\ac\manchester\spinnaker\data_spec\Functions.java

What is the problem you see?
these all define the same constant (albeit some with naming little differences, but some are constant names), with the same value

What do you think it should do instead?
its likely some (im not silly enough to assume all can) can be merged into one constant located in constants.java

What version of our software are you using?
master

dkfellows, 2018-12-06:
No. They don't represent the same thing.

alan-stokes, 2018-12-06:
They do in ReadFPGARegister and WriteFPGARegister for starters. So that one at least needs to be merged

dkfellows, 2018-12-07:
One represents the register mask for a read and the other for a write. _Those are not required to be the same._ Yes, this is all shonky bit bashing, but no, I **refuse** to merge them.

closed 2018-12-07 by dkfellows

alan-stokes, 2018-12-07:
ok, i will relent. But rename them so that they dont have the saem name. Such that we dont get motivated to assume they cant be merged