SpiNNakerManchester/JavaSpiNNaker #183: ChipMemoryIO.java

alan-stokes, 2018-12-06:
Full path to where you find the problem:
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\io\ChipMemoryIO.java
and equiv in python
C:\Users\alan\Documents\software\SpiNNMan\spinnman\utilities\io\memory_io.py

What is the problem you see?
	// TODO What is this value _really?_
	private static final int SDRAM_START = 0x60000000;

What do you think it should do instead?
this is the unbufferedSDRAM, not buffered which is 70000000; maybe a rename is better. (or visa versa)
also the //TODO might want to be removed 

What version of our software are you using?
master

rowleya, 2018-12-07:
Buffered SDRAM means that writes go through a write buffer.  Unbuffered means that they go directly to SDRAM.  Reads are unaffected in general.  If you are writing lots of data, it is unlikely to matter much since the write buffer is limited in size.  Here you probably want to use Unbuffered anyway, as it will then block until the write is definitely done (as I understand it).  Using Buffered writing means that the write may or may not have happened at the time of the response.

alan-stokes, 2018-12-07:
yep, but for clarity sake, naming them as SDRAM_UNBUFFERED_START would mean poor souls coming to this dry in future can know there's a difference.

dkfellows, 2018-12-07:
I prefer `UNBUFFERED_SDRAM_START`

alan-stokes, 2018-12-07:
just to add. i learnt this from @lplana  when i was looking at nengo, going what the heck is it doing at 60000000 when i was sure our sdram's tarted at 70000000 (we must have changed this at some point in the 4 years). So that lesson got hit, after 4 years! so given it hit me, i want it exposed as a thing in the code, so new peeps dont find this secret lateron

Christian-B, 2019-01-02:
fixed in both Java and Python so ok to closed.

closed 2019-01-02 by Christian-B