SpiNNakerManchester/JavaSpiNNaker #191: send_scp_request_to

alan-stokes, 2018-12-07:
Full path to where you find the problem:
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\transceiver\Transceiver.java

What is the problem you see?
		 * update the SCAMP connections replacing any x and y with the default
		 * SCP request params with the boot chip coordinates
		 */
		for (SCPConnection sc : scampConnections) {
			if (sc.getChip().equals(DEFAULT_DESTINATION)) {
				sc.setChip(machine.boot);
			}
		}

What do you think it should do instead?
Perfection is achieved, not when there is nothing more to add, but when there is nothing left to take away.

What version of our software are you using?
master

linked to https://github.com/SpiNNakerManchester/SpiNNMan/issues/149

dkfellows, 2018-12-07:
I've no idea what to do instead.

alan-stokes, 2018-12-07:
just remove it. its redundant code. as rowley and me been debating a bit of the afternoon with food fog. @rowleya to confirm

Christian-B, 2019-01-02:
closing as Python side covered by another issue and assigned to @rowleya

closed 2019-01-02 by Christian-B