SpiNNakerManchester/JavaSpiNNaker #193: default chip

alan-stokes, 2018-12-07:
Full path to where you find the problem:
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\connections\SCPConnection.java

What is the problem you see?
the DEFAULT_CHIP is risky for causing multiple connections to point at the same boot. forcing the app to set it will remove this risk. On most of the use cases, we already set the chip. only place that seems to not use it is line 414 in txrx which is:

		if (scampConnections.isEmpty()) {
			connections.add(new SCPConnection(host));
		}
in the consturctor

What do you think it should do instead?
above

What version of our software are you using?
master

linked to:
https://github.com/SpiNNakerManchester/SpiNNMan/issues/148

dkfellows, 2018-12-10:
I'll figure out what to do about this once I see how you guys fix SpiNNMan…

dkfellows, 2023-02-17:
Using the `getTransceiver()` method of the new spalloc client job class (in comms) results in a transceiver with separate connections to each board (and no BMP access at all).

closed 2023-02-17 by dkfellows

dkfellows, 2023-02-17:
If you do it all manually, you need to get things right. Current preferred mode of operation is you ask spalloc (old or new) to give you a transceiver; it knows how. Only doesn't really apply when you use local boards, and then you can either use the defaults (fine for a single board) or Get It Right&trade;.

commit b0e5ac66 (2023-02-17): Give the old spalloc client API the ability to make a transceiver

Effectively fixes #193 for all but the smallest of setups, where the
choices of defaults actually work.
PR #734 (2023-02-17): Give the old spalloc client API the ability to make a transceiver
Effectively fixes #193 for all but the smallest of setups, where the choices of defaults actually work.