SpiNNakerManchester/JavaSpiNNaker #194: mutible

alan-stokes, 2018-12-07:
Full path to where you find the problem:
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-machine\src\main\java\uk\ac\manchester\spinnaker\machine\RoutingEntry.java

What is the problem you see?
    /**
     * Removes a link ID/Direction if it existed.
     *
     * @param oldValue The Id of the processor to remove.
     * @return {@code true} If this entry contained the specified
     *      link/direction.
     */
    public boolean removeLinkID(Direction oldValue) {
        return linkIDs.remove(oldValue);
    }

What do you think it should do instead?
this makes the class mutable. Which is against our mantra about spinnMachine. Its only ever used by 1 test, which makes it even more stinky. 

remove the test and the method

What version of our software are you using?
master

Christian-B, 2018-12-07:
This issue is incomplete!

Removing only one of two remove methods and leaving the public add methods does not change the Object Immutable.

Christian-B, 2018-12-07:
Mutibilty was a left over from when the Fiddler removed bad links.

closed 2018-12-10 by dkfellows