SpiNNakerManchester/JavaSpiNNaker #204: redundant method

alan-stokes, 2018-12-10:
Full path to where you find the problem:
C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\connections\UDPConnection.java

What is the problem you see?
public DatagramPacket receiveWithAddress()

What do you think it should do instead?
never used, maybe go bye bye

What version of our software are you using?
more-fixing

alan-stokes, 2018-12-10:
same for sendTo 
public void sendTo(DatagramPacket data, InetAddress address, int port)

alan-stokes, 2018-12-10:
public void sendTo(byte[] data, InetAddress address, int port)

alan-stokes, 2018-12-10:
same for FileIO.java (C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\io\FileIO.java) for constructor:

public FileIO(File file, int startOffset, int endOffset)
			throws IOException {
		this(new RandomAccessFile(file, "rw"), startOffset, endOffset);
	}

its worth noting that the ide thinks the entire class isnt used............. confused

alan-stokes, 2018-12-10:
same issue for MemoryIO (C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\io\MemoryIO.java) it thinks the class is never used

alan-stokes, 2018-12-10:
same for BootMessages.java (C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\messages\boot\BootMessages.java)

public BootMessages(Map<SystemVariableDefinition, Object> extraBootValues) {
		this((SystemVariableBootValues) null, extraBootValues);
	}

alan-stokes, 2018-12-10:
GetMachineProcess (C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\transceiver\processes\GetMachineProcess.java) 

	public Map<ChipLocation, ChipSummaryInfo> getChipInfo() {
		return unmodifiableMap(chipInfo);
	}

alan-stokes, 2018-12-10:
thus why i put the question marker on it. im not sure if some of these are stuff we actually don't want to keep around. given the comms was modified quite a bit to fit with java, or if its jsut part of the api. Theres a few places where the ide flaggs it, but i can tell its part of the api. but these im not sure.  

for example, the send_to in UDPConnection, in python this is only used by a test and the utility_functions.send_port_trigger_message method, which i havent come across in the java (or at least, it doesnt ring bells in my head, and the ide cant find it....) I know in python, its to pop a hole in the firewall, but in the java......?????

it also is used by SCAMPConnection send_scp_request_to, but that method itself is never used...... so some of this could be legacy code in python that should be pruned as well.

rowleya, 2018-12-10:
Yes, this is down to the question of how much you want to keep as it makes sense as an object vs. how much you use.  I guess if it is complex functionality, I would agree that the maintenance (and creation) probably makes it not worth it.  For small one-liner functions though, having the ability at some point to do this with virtually no overhead is nice.

alan-stokes, 2018-12-10:
oh i completely agree. which is why i flagged these. to allow us to decide if we want these or not. I'd like to know how the java gets around the firewall poking issue. lol. As after that reply, i still cant find it. lol. Maybe @dkfellows  has some ideas

dkfellows, 2018-12-10:
The only time we do anything in Java related to receiving and wanting the address received from is in `LocateConnectedMachineIPAddress`, _but_ FEC also uses the capability in `spinn_front_end_common.utilities.database.DatabaseConnection` and that's an area that hasn't been looked at to see whether it should be converted or not.

closed 2023-02-24 by dkfellows