SpiNNakerManchester/JavaSpiNNaker #206: defaultChip vs BootChip

alan-stokes, 2018-12-10:
Full path to where you find the problem:
starts in SCPRequest (C:\Users\alan\Documents\software\JavaSpiNNaker\SpiNNaker-comms\src\main\java\uk\ac\manchester\spinnaker\messages\scp\SCPRequest.java)

What is the problem you see?
	public static final CoreLocation DEFAULT_MONITOR_CORE =
			new CoreLocation(DEFAULT_DEST_X, DEFAULT_DEST_Y, 0);
    /**
	 * The pseudo-location of the default chip.
	 */
	public static final ChipLocation DEFAULT_CHIP =
			DEFAULT_MONITOR_CORE.asChipLocation();

What do you think it should do instead?
the 255 255 is actually the boot chip, not the default chip. took a good while with @lplana and @rowleya to get to that logic, but basically, default_chip is a misnomer

What version of our software are you using?
more-fixing

dkfellows, 2018-12-10:
I see what you're saying, but I don't know what to do about it. 😄

alan-stokes, 2018-12-10:
I think its a renaming task.

PR #212 (2018-12-14): Yet another tranche of small fixes
fixes #112 
fixes #167 
fixes #168 
fixes #169 
fixes #200 
fixes #201 
fixes #202
fixes #203 
fixes #206
closed 2018-12-10 by dkfellows