SpiNNakerManchester/JavaSpiNNaker #216: TestTransceiver fails in windows

Christian-B, 2018-12-20:
Full path to where you find the problem:
uk.ac.manchester.spinnaker.transceiver.TestTransceiver (in tests of course)

What is the problem you see?
java.net.SocketException: Network is unreachable: connect
	at uk.ac.manchester.spinnaker.transceiver.TestTransceiver.testSetWatchdog(TestTransceiver.java:191)

What do you think it should do instead?
possible check for window or skip on faik

What version of our software are you using?
extractor2

dkfellows, 2019-04-15:
That one is supposed to be run with a mock against an unreachable host. (It also works for me when I re-enable the test.) I wonder what's going wrong.

dkfellows, 2019-04-15:
Looks like the same root cause as #215, and I suspect the problem is an ISP being “helpful” because “all the world ever runs is webpages”.

closed 2019-04-15 by dkfellows