SpiNNakerManchester/JavaSpiNNaker #217: Java DSE depends on startaddress being cleared

Christian-B, 2019-01-02:
Full path to where you find the problem:
unsure somewhere in the HostExecuteDataSpecification

What is the problem you see?
DSE fails if the field  startaddress is not cleared.
Python however does not check and writes all

What do you think it should do instead?
Write all where content is not NULL

What version of our software are you using?
extractor2

Note: there is currently a Python side work around where HostExecuteDataSpecification calls dw_write_info.clear_write_info()

dkfellows, 2019-04-09:
I'm sorry, but I don't understand what the problem is.

dkfellows, 2019-04-09:
Ah, found it. Need you to specify what should happen in this case, but doing a reset is pretty easy. The main issue is that we don't guarantee that a reload will use the same section of memory. It _might,_ but it isn't a guarantee. 

https://github.com/SpiNNakerManchester/JavaSpiNNaker/blob/1dac2b4274f49f92bb9f0f46fce96f790a7fc954/SpiNNaker-storage/src/main/java/uk/ac/manchester/spinnaker/storage/sqlite/SQL.java#L128

dkfellows, 2019-04-15:
Do you want this fixed?

PR #228 (2019-05-07): Ignore startAddress when reading specs from DB
Remove support for restarting the evaluation and uploading of data specs; we never used it, and it just caused confusion in the Python side of things.

fixes #217
commit 0a4e9425 (2019-05-07): fixes #217

Remove support for restarting the evaluation and uploading of data
specs; we never used it, and it just caused confusion in the Python side
of things.
closed 2019-05-07 by dkfellows