SpiNNakerManchester/JavaSpiNNaker #218: Gather fails!

Christian-B, 2019-01-02:
Full path to where you find the problem:
uk.ac.manchester.spinnaker.front_end.interfaces.buffer_managementDataReceiverRunner.gather(...
Or the classes it calls

What is the problem you see?
10:49:43 DEBUG UDPConnection:364 - sending data of length 27 to /192.168.240.253:17893
10:49:43 DEBUG UDPConnection:368 - message data: [0, 0, 87, ff, 0, ff, 1, 1, ff, ff, 3, 0, 64, 0, 43, 7f, 0, f5, 1, 0, 0, 0, 0, 0, 0, 0, 0]
10:49:43 DEBUG UDPConnection:375 - sent 27 bytes
10:49:43 DEBUG SCPRequestPipeline:379 - waiting for message...
10:49:43 DEBUG UDPConnection:509 - timout on UDP(/192.168.240.254:59368 <--> /192.168.240.253:17893) will happen at 1000 (1)
10:49:43 DEBUG UDPConnection:520 - wait:0:true:true
10:49:43 DEBUG SCPRequestPipeline:381 - received message RC_OK
10:49:43 WARN  DataGatherer:333 - problem when downloading a board's data
uk.ac.manchester.spinnaker.front_end.interfaces.buffer_management.DataGatherer$FullFailureException
	at uk.ac.manchester.spinnaker.front_end.interfaces.buffer_management.DataGatherer$Downloader.doDownload(DataGatherer.java:576)
	at uk.ac.manchester.spinnaker.front_end.interfaces.buffer_management.DataGatherer.handleOneRecordingRegion(DataGatherer.java:426)
	at uk.ac.manchester.spinnaker.front_end.interfaces.buffer_management.DataGatherer.doDownloads(DataGatherer.java:369)
	at uk.ac.manchester.spinnaker.front_end.interfaces.buffer_management.DataGatherer.downloadBoard(DataGatherer.java:318)
	at uk.ac.manchester.spinnaker.front_end.interfaces.buffer_management.DataGatherer.lambda$addTask$0(DataGatherer.java:163)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

What do you think it should do instead?
Not timeout

What version of our software are you using?
Extractor2

dkfellows, 2019-01-05:
Current state: the messages just aren't being received. I believe I am now programming the iptags correctly (but must confirm that) so where are they going?

Christian-B, 2019-01-07:
Houston we have liftoff

Christian-B, 2019-01-07:
Works consistently with voltage of up to  37,000 worrds
1000 timestamps for 36 neurons

But fails at 38,000 words
1000 timestamps for 37 neurons
as well as 2000 timestemps for 18 neurons.
(remember there is also a word for timestamp)
java.nio.BufferOverflowException
	at java.nio.Buffer.nextPutIndex(Buffer.java:527)
	at java.nio.HeapByteBuffer.putInt(HeapByteBuffer.java:372)
	at uk.ac.manchester.spinnaker.front_end.interfaces.buffer_management.MissingSequenceNumbersMessage.create(MissingSequenceNumbersMessage.java:119)

Christian-B, 2019-01-11:
testing continues but this appears to be working,

dkfellows, 2019-01-11:
Current gatherer works on one board. It'd work across many boards if we actually used a transceiver that knew about the other boards, the information about which has been “cleverly” thrown away…

Christian-B, 2019-01-11:
it also always fails in python2.7 due to python 3 syntax in the line:
params = [self._java_call, *each_param, '-jar', self._jar_file]

Christian-B, 2019-01-14:
python2.7 resolved using List concatenation.

Christian-B, 2019-01-14:
First tests using multiple boards are passing.

We had a case of a problem when the last empty package needed to be resubmitted but @dkfellows has done a fix so fingers crossed.

testing continues based on a combination of this and the Jenkins branches.

Christian-B, 2019-01-14:
sPyNNaker8/p8_integration_tests/test_various now all pass.
Or fail for clearly non Java issues.

closed 2019-01-14 by Christian-B

Christian-B, 2019-01-14:
All integration tests up to sPyNNaker8/p8_integration_tests/test_energy_params_validity
work or at least fail for non Java reasons