SpiNNakerManchester/JavaSpiNNaker #338: Identify users

dkfellows, 2021-06-21:
For #325, how should we identify users? How should we register users?

Old spalloc has no security at all. Everything is by self-asserted identity. Fortunately we've not had problems with bad actors.

Christian-B, 2021-06-21:
ideally for Jupiter and HPB users we should use the id they use to log into that level.

So that leaves just the direct spalloc users which is limited to users who are inside the Manchester University network, so the risk of none accidental bad actors is lower. So for now I think we can accept the spalloc_user cfg value.

if we need a level of security (and I dont think we do) I would have a systems that users enter an email address the system generates a private keys and emails it to the user.
then the users have to use a know private key,

Christian-B, 2021-06-21:
As this is linked to https://github.com/SpiNNakerManchester/JavaSpiNNaker/issues/342 I would agree we need either a private key or a check that the user requesting to view or change stuff is inside the Manchester Uni firewall.

Christian-B, 2021-06-21:
If we can verify HBP credentials then we could say that spalloc users have to use their HBP id as their "spalloc_user" as otherwise they will not be able to access their data or jobs via the server.

rowleya, 2021-06-21:
A thought is to have a "default" authentication system which might be OAUTH based e.g. EBRAINS users (this would likely need to be done through Bearer tokens as spalloc will likely be accessed by API rather than user interface).  There might then be a secondary authentication for local users that then don't use the default option e.g. IP address + local username, or even just UMAN (LDAP) authentication.  If there is a "local username" option, that implies having to have an interface to register users (or allow users to self-register if desired).

Christian-B, 2021-06-21:
All local users have or can easily get an ebrains account so no need for a dual systems,

The testing jobs, data ect should only be accessed by admins so as long as the iobs run all other access can be via an admin account rather than oweber account.

rowleya, 2021-06-21:
EBRAINS login is OAUTH which normally means that a browser is required to log in.  The user can then get a Bearer token which can be used with APIs, where the API then has a way to confirm who the user is using that token.  This is reasonably easy when the user is behind a service that is accessed on the web i.e. Jupyter or the NMPI portal system (though even then, this will require refresh tokens to keep the Bearer token alive).  If we are going to also rely on this for non-web users, we would need to provide some way to get a non-bearer token e.g. an API token.  This could be done with a one-off request, with the user then storing the token in their config file to access spalloc.

Christian-B, 2021-06-21:
For spalloc we can in step 1 trust that the the provided spalloc_user is their ebrains user id.  If they provide and incorrect id then too bad they will nto be able to access that job via the API.

dkfellows, 2021-06-21:
We can support multiple login mechanisms. (There'll also need to be session tokens and stuff like that, but I believe Spring Security will handle most of the details there.)

dkfellows, 2021-06-28:
[This](https://www.baeldung.com/spring-security-openid-connect) seems at least worth reading carefully, as it is a tutorial that handles being an OpenID “client” (for Google, but hopefully HBP isn't too different).

dkfellows, 2021-06-28:
Also [see this](https://wiki.ebrains.eu/bin/view/Collabs/collaboratory-community-apps/Community%20App%20Developer%20Guide/1.%20Registering%20an%20OIDC%20client/) for obtaining the API key and secret for contacting EBRAINS for OIDC (OpenID Connect).

dkfellows, 2021-06-30:
There's now a table in the database with user information. I intend for users with NULL passwords to have to log in via OpenID (and we can add them to the DB on demand).

dkfellows, 2021-12-17:
This is now sorted; HBP users don't need to register.

closed 2021-12-17 by dkfellows