SpiNNakerManchester/JavaSpiNNaker #340: Enable and disable board allocation

dkfellows, 2021-06-21:
There should be (in #325) an easy way to enable and disable a board (e.g., because it has a fault or something). The data model already supports this; there is a column of the `boards` table that says explicitly whether a board is believed to be functioning (and a derived column that says whether the board may be allocated; that also depends on whether it is already in use).

I can put this in the general admin interface that I've started writing.

dkfellows, 2021-06-21:
When enabling or disabling a board, should it be done by IP address, (X,Y,Z) triad logical coordinates, or (cabinet,frame,board) physical coordinates? Or all of them? (I can also use the database's own internal identifier, but that has literally no other meaning at all.)

rowleya, 2021-06-21:
For a user, IP address is likely to be easiest, though some more advanced uses might prefer cabinet, frame, board.  I don't think anyone works in triad coordinates except spalloc itself.

rowleya, 2021-06-21:
Note that if the board was to be disabled after finding an issue during a job, it could also be by job id if that were possible.

Christian-B, 2021-06-21:
As our reports of bad boards tend to be ip address based that is most usefull.

Christian-B, 2021-06-21:
A nice to have but not critical is where there is an easy to access ai where the boards are temporarily disabled suggest to a review by an admin.
We could then have Jenkins disable boards it has an issue with even if the issue was really a network disconnect and not a board issue.

dkfellows, 2021-06-21:
That would tie into the set of work that #285 is part of, and I would approve of it a lot. However, that's an operation that we might want to partially restrict by user. Hmm…

rowleya, 2021-06-21:
I guess an option would be a voting system.  If enough people (enough being defined in a config) report the same error, the board will be disabled until reviewed.

dkfellows, 2021-06-28:
Admin API now supports enabling and disabling a board. (Boards may be identified by triad triple, by cabinet-frame-board physical triple, or by IP address.)

Still need mechanism for users to report issues with a board.

dkfellows, 2021-12-19:
Issue reporting mechanism is there, but not in the GUI. Two (configurable) reports for a board results in the board being taken out of service.

We perhaps ought to self-report boards that we cannot turn on? Hmm...

rowleya, 2021-12-20:
I think the idea is for this to be at least somewhat automatic, though GUI access would be nice to re-enable boards in particular.  Disabling boards that have power issues is definitely a good idea!

dkfellows, 2022-01-29:
Board control UI needs more work. Much more. The problem with what we have now is that we are confused between board lookup and board state adjustment.

dkfellows, 2022-02-07:
We can now reliably disable and enable boards, and even entire machines, both via the administration interface. See #417