SpiNNakerManchester/JavaSpiNNaker #342: Security model

dkfellows, 2021-06-21:
If we address #338, how will this affect the information returned?

My feeling is that owner, creation request and keepalive host information should only be returned to admins (about jobs that they don't own). Only owners and admins should be able to change a job (including keeping it alive) but general users _including unauthenticated users_ should otherwise be able to read the machines and jobs.

----

The restricted values should be restricted because of PII issues.

Christian-B, 2021-06-21:
I would agree that information such as how many jobs are running, how big they are in total can be made available to all users without giving any privacy info away especially as this is likely to include the testing platform.

I even dont see an issue if normal users can see limited info on each others jobs.  I would not implement a systems so that they can see this data, but also not worry if through stuff that is there they could work it out.  What is not cool is if they can see exactly what is running or the data.

I would prefer is user X does not get the id for users Y jobs (where ID is what would be used to view or change the job)

At this stage I would NOT implement a data sharing function.  If user x wants to shared data with none admin user y they have to pull it off the machine.

I also don't see any issues with everyone being able to see stats on how many jobs have been run for how long on how many boards ect even if the stats here split testing vs none testing.

Christian-B, 2021-06-21:
I would also make it a policy that we reserve the right to make public high level job info on user id (for example HBP ld) basis.

rowleya, 2021-06-21:
Above seems reasonable; ideally it would be useful for any user to see what is in use, but agree that they don't need to know who is using it.  Admins should probably be able to see who is using what though.

dkfellows, 2021-06-21:
I'd restrict information visibility by standard PII rules (we'll probably have to do that anyway, as a legal requirement). Non-read operations exposed will be generally this:
* Logged in users can create jobs. They will own the jobs they create.
* Job owners can manipulate their own jobs.
* Admins can manipulate everything. They can also list dead jobs. (We never _really_ delete them.)

dkfellows, 2021-06-28:
# Authorization Model
I now have three roles (and by implication a fourth):

* `READER` — can read the machines' configuration and summary job information
* `USER` — can read job details and create+manipulate jobs.
* `ADMIN` — can do everything, including use the admin interface to import new machine definitions. Only admins can impersonate other users.
* The fourth role is where the login is authenticated but there's no role assignment: only the service version can be obtained. (This might be useful for health-check services.)

In addition, some job fields are shrouded when read by any non-admin other than their owner, and only the owner and admins can change an existing job (including deleting it).

# Authentication Model
I _really_ need help with authentication! What I've got right now for that is a placeholder only (single user, password and roles set directly in code) that I've used for testing. What I'd _like_ to have is support for both local service accounts (for ourselves) and delegating to whatever auth scheme is used in HBP.

rowleya, 2021-06-28:
It sounds like a READER is an unauthenticated USER i.e. people can see that there are jobs and machines without details without authentication.  Is that correct?

Then a USER can read only their own job details and can only manipulate their own jobs.  Again, just checking!

A thought on authentication would be that when a user first logs in (i.e. no role assigned in the database), they will be automatically assigned the role USER.  This can be associated with their EBRAINS username; thus usernames should be globally unique over any authentication mechanism.  It should then be possible for an ADMIN user to change that role.

Note this means that the authentication mechanism and the role database can be separated.  So then any authentication mechanism can be added at any point.  Only after authentication has been achieved will the role database be looked at, with the only information passing from authentication to authorization being the username.  The code can then assume a username that it receives has been authenticated.

Adding EBRAINS authentication is then a slightly tricky step for an API, mainly because of the OpenID Connect process that is in place.  The process that needs to be followed is therefore a) user logs in to EBRAINS *using a web browser*.  This results in a BEARER token being produced (this probably needs our own simple web GUI to display the user BEARER token to them to allow them to copy it).  User then uses BEARER token to speak to Spalloc.  On the spalloc side, it then "simply" (not quite sure of the details) needs to speak to the EBRAINS systems to validate any BEARER token.  Additionally, it may have to keep a refresh token and keep pinging the service periodically to ensure the BEARER token is kept alive.

An alternative to the above is that the BEARER token is used to obtain a permanent (or semi-permanent) spalloc API token.  When the token is presented, this authenticates the user.  This is similar to the above except that the refresh token is then not needed.  This is how the EBRAINS drive appears to operate.

Christian-B, 2021-06-29:
I like the idea of a spalloc API token.  

The uses allowed by the spalloc token could be an additional role that of RUNNER
With a spalloc token you can start and use the current spalloc job and obtain data from that job.
This avoids the need for full OpenID authentication every time you run a spalloc job

However for access to previous jobs and history you would need the full OpenID authentication  to become a USER.

Christian-B, 2021-06-29:
In Python you can obtain READ ONLY access to an sqllite database so I assume you can in Java as well.

We could therefor have the ADMIN user script access be the ability to run any SQL query on the database with reduced risk as they would not be able to even accidentally damage the data,

rowleya, 2021-06-29:
My thought was that the API token would be directly associated with the user, and would therefore give the user access as if they had logged in directly.  This would be like a github personal access token (see https://docs.github.com/en/github/authenticating-to-github/keeping-your-account-and-data-secure/creating-a-personal-access-token).  

This could be further secured with a difference between a USER logged in directly now, and a USER logged in with a token, which might be done with roles e.g. USER and USER_TOKEN (not sure about if this could work since they would really be referring to the same user; maybe something like the restrictions that github provides for tokens).  The extra security would simply be to disallow token creation or deletion when USER_TOKEN was the role (and maybe stop password changing as well if that is a thing that can be done).  That would then mean that a hypothetical "malicious user" who got hold of a token that wasn't theirs wouldn't be able to use it to stop the real user getting in.

In the same vein, an ADMIN user logged in with a token wouldn't have admin privileges.

I am saying all of the above without any idea of how easy this is to implement... so feel free to complain that this is going to be too hard to manage!

dkfellows, 2021-07-12:
**Note:** If adding any new pages, pay particular attention to the [CSRF prevention tokens](https://docs.spring.io/spring-security/site/docs/5.0.x/reference/html/csrf.html). That's an area which is _great_ for mysterious “Computer Says No” errors.

dkfellows, 2021-07-14:
I believe I've got the local auth stuff all working right. Logging in as an admin I get extra links to if I log in as a normal user. (I've not yet put any jobs in to test that part, but I _think_ that's working too.) The links are also checked for access; if you're not told a link, you can't have it. There's also session binding and CSRF protection in place. Logged in users can see _some_ details of all jobs, but not all. (They can, of course, see all details of their own jobs, and admins can see all details of all jobs.) I've got three levels of logged-in protection right now:

0. Can see basic service status (the `BASIC` permission; all logged in users get this).
1. Can see machine and job information (this is what `READER` permission/trust level implies)
2. Can submit jobs (this is what `USER` permission/trust level implies, as well as what `READER` implies).

Passwords are now encrypted within the database correctly (by enhancing the DB with extra functions) and conceptually don't leave. (They do actually, but only because the functions are implemented in Java; avoiding that would be horrible. But the functions are the only things that touch the password encoder.)

API access is secured in the same fashion as the UI (with equivalent protection levels). I'll have to write the client as there's some critical low-level details that need to be got right in the session management, and I'm at least aware of their implications. 😁  I've yet to finish building out the UI — the machine and job detailed views aren't there — but for machine and job _lists_ it's probably good enough to just tell the page to refresh every 30 seconds. (API users can do long queries instead.) Permissions modulate what information is actually given to build those pages; that which the current user is not allowed to see will never reach the page renderer, let alone get transmitted to the actual user.

The only part that's missing in the auth picture is non-local auth, i.e., using OIDC with EBRAINS/HBP. I don't understand the details there, but I plan to give all ordinary HPB users the `USER` permission level.

dkfellows, 2021-11-25:
#380 has made it so that we now run (apparently) on the standard HTTPS port.

dkfellows, 2021-12-17:
Logging in with HBP/EBRAINS is now supported, as are bearer tokens (provided they're issued by the HBP/EBRAINS keycloak service). Security model is done.

closed 2021-12-17 by dkfellows