SpiNNakerManchester/JavaSpiNNaker #348: Web interface for viewing status

dkfellows, 2021-07-05:
It'd be really nice to have some way to do things with spalloc server over the web.

I'd like the general capabilities of `spalloc-ps` and `spalloc-machine` to be available (plus some of what `spalloc-where-is` does, and I'd like that to be in a web page.

# From the spalloc (client) docs

[image]
[image]

Note that showing user info will have to be a logged-in only thing, we can use distorted hexagons to give a better view of the machine, and we can present extra information on mouse-over.

Christian-B, 2021-07-05:
I would highly recommend doing each if these in layered methods.

First layer to get the data as a java object which is also responsible for making sure the caller has the rights.

Second layer to convert the java object(s) in to html, json ect.

dkfellows, 2021-07-05:
I've already got the information surfaced from DB to Java and Java to JSON-based ReST API, all with appropriate permissions applied. This issue is about providing a way to view the information in a web page.

dkfellows, 2021-11-18:
This now largely works. It's probably still buggy, but at least up to the scale of a 24-board test deployment it's fine. One of the nice things is to get a view of a machine, see an allocation on the machine map, and then click on it and go to the page describing the allocation in detail. (You can only do that for jobs you're authorized to see the details of, but that's going to be so nice for admin.)

closed 2021-11-18 by dkfellows