SpiNNakerManchester/JavaSpiNNaker #366: Keep job details beyond the end of the job

rowleya, 2021-09-22:
For debugging, it could be useful to keep job details beyond a job being "destroyed".  This could then let users "see" the boards that were used in a job in the past.  This might have an expiry date e.g. 1 month.

dkfellows, 2021-11-18:
Job details are kept (first in operational store, and then gradually migrated to longer term append-only storage in a separate database). But how should they be displayed to users? Note that some users (e.g., key service users) may end up having _very_ large numbers of old jobs. (The destroyed jobs should be listable right now — excluding the archived jobs — _except_ I've literally not written the UI to do it. The REST API supports paged access.)

rowleya, 2021-11-19:
If this only supports jobs in the operational store, that would be fine as it is more likely that it would be used to find out what went wrong on something reasonably recent (and with the knowledge that the jobs will disappear eventually, debugging information can be extracted early and logged elsewhere if the issue isn't to be fixed immediately).

dkfellows, 2021-11-25:
The tombstoning copies old jobs (currently configured to be ones 14 days dead) to a database that I'm treating as if it is append-only.

dkfellows, 2021-12-19:
What I don't currently keep is the list of boards (by id most likely) that were allocated to the job...

rowleya, 2021-12-20:
I guess it is possible to calculate that if you know where it starts and how many boards, but it would be nice to have anyway.

dkfellows, 2022-03-17:
We should now keep the allocation records beyond the end of the job; they'll also get migrated to the historical data DB when the job gets moved.

Keeping this issue open: not sure that that's all the information we want to keep.

closed 2023-02-24 by dkfellows