SpiNNakerManchester/JavaSpiNNaker #413: Group / Collab quotas

rowleya, 2022-02-02:
The HBP model of quota allocation assigns quota to collabs rather than individuals.  To make synchronization of quotas between the HBP service and the SpiNNaker allocation easier to manage, it would be helpful if the SpiNNaker allocation service could allocate quota to groups.  

Thoughts:
 - A user would need to be identified to be authorized to use quota from one or more groups.
 - A user can still have a default personal quota (which might be 0).
 - As part of a job, a user can identify which quota a job should be allocated from.  If the user is not authorized to use the quota, a quota of 0 can be assumed.  If the user doesn't identify a quota, their personal quota can be used by default.

rowleya, 2022-02-08:
Looks like I might have it backwards:
https://wiki.ebrains.eu/bin/view/Collabs/the-collaboratory/Documentation%20IAM/FAQ/OIDC%20Clients%20explained/Authenticating%20with%20your%20OIDC%20client%20and%20fetching%20collab%20user%20info/

dkfellows, 2022-02-08:
Hmph. Looks like we should get both `group` and `team` and poke around in what comes back.

dkfellows, 2022-02-08:
The `unit` from the `userinfo` request also looks relevant.

dkfellows, 2022-02-08:
This is why I _hate_ dealing with Keycloak and OIDC; there's a colossal amount of flexibility, such that pinning down how things work in our specific case is really very difficult indeed.

rowleya, 2022-02-08:
Yes, it is a generic solution to which there has been applied a custom model for this use case unfortunately.  Note that those docs above are for EBRAINS specifically.

dkfellows, 2022-02-08:
That means that job submissions will need to identify which group to account against… at least unless a user only has access to a single group.

rowleya, 2022-02-08:
Yes that sounds about right all round.  A user can be in many groups, but a job will only take quota from one of those groups.  If we have a default group for each user which only contains the user (like linux ;), if a user doesn't specify the group, they would then get that default group.  The default group would also probably have either 0 quota by default or some fixed value we let everyone have.

Christian-B, 2022-02-08:
Would a Job be directly related to a group rather than to a user?

If so when is a Job related to a user and when to a group?

And if a user in many groups starts a job which group does it go into?

dkfellows, 2022-02-08:
Auto-inflating groups can be done too.

In any case, the big challenges here are identifying the groups from Keycloak, designing the fundamental data model, and working out what to do about migrating our existing stuff. (The data model changes currently look rather like breaking changes…)

rowleya, 2022-02-08:
I think we can stick with a job being related to a user.  If a user is in many groups, they have to specify which group the quota for a job comes from.

dkfellows, 2022-02-08:
In terms of the database model, the job has to be associated with something that owns a quota (which sounds like a group in our model). We can pick a default, we can inflate something automatically, but we _can't_ leave it just open; that'd be pushing far too much complexity at the accounting side of things.

Christian-B, 2022-02-08:
I assume that the group a job is related too will also affect who has access to the job.

In which case it make sense for the user to have to say which group to assign the job too.

dkfellows, 2022-03-10:
The biggest problem with this right now is that a typical HBP user (well, me) has about 90 groups that they are a member of. That's one whole huge lot of groups! That basically makes it essential for users to specify which group (especially which collabratory) they are submitting a job against when they submit it. The basic API for doing that is there, but I've no idea what that's going to look like when surfaced further up the stack, from the perspective of Python code.

Still, I now consider this to be a problem for a different issue; what we've got _works_.

PR #419 (2022-03-04): Spalloc server groups
Implementation of groups for spalloc server (which will fix #413). Groups are the units to which accounting is performed, especially within the context of HBP.

- [x] Core model changes
- [x] Database migration (or rebuild from scratch)
- [x] Link jobs so their accounting goes against their group
- [x] Create default group for local users
- [x] Inflate groups from OIDC responses
- [x] UI for group discovery and quota control
- [x] Admin-level API for quota usage export
closed 2022-03-10 by dkfellows