SpiNNakerManchester/JavaSpiNNaker #414: Quota API

rowleya, 2022-02-02:
Related to #413 it would be useful to have an API with which an external script could interact to synchronize quotas between HBP and SpiNNaker allocations.

Things this might need to do:
 - Add a group and allocate users to the group with a group quota.
 - Read the group quotas remaining.

dkfellows, 2022-02-05:
Putting a REST API in is easy enough (provided we have a good definition of who can do what operations) subject to the issues relating to the naming of things.

dkfellows, 2023-02-24:
This is the main part left undecided. I simply don't know what APIs elsewhere we would be interacting with. But the basic units of accounting would be board-seconds; that's totally what it is easy for us to work with.

dkfellows, 2023-06-29:
See:
* #929

rowleya, 2023-07-19:
This has been done otherwise so closing.

closed 2023-07-19 by rowleya